spacelab-ufsc/battery-module-4c #4: Hardware: Review mechanical attachment method

andrempmattos, 2020-06-28:
The mechanical attachment for a plastic/polymer cover/wrap 

Some ideas:

YanAzer, 2021-02-10:
Some new images for inspiration:
[image]
[image]
https://www.google.com/imgres?imgurl=x-raw-image:///6634c2ea12cae2710d74224933948a37fe1926551f1a21b6ea2360aa884c3e4f&imgrefurl=http://mstl.atl.calpoly.edu/~workshop/archive/2018/Spring/Day%25201/Session%25202/NathanEngland.pdf&tbnid=ceLKEurCNGZkVM&vet=1&docid=Pg2TiIZvuWyThM&w=666&h=435&source=sh/x/im
https://www.google.com/imgres?imgurl=http://www.imtsrl.it/_media/img/large/battery-pack-3-1024x682.jpg&imgrefurl=http://www.imtsrl.it/battery-pack.html&tbnid=6zWjY_rsfmTN3M&vet=1&docid=XPJP-t2zuxd0rM&w=1024&h=682&source=sh/x/im

YanAzer, 2021-04-28:
@spacelab-ufsc/mechanical some new ideas for the new mechanical case:

1) Add new mouting holes between the batteries, this will require them to be a bit por spaced.

[image]

2) Add a "tab" on the sides of the pcb to be enable placing the required mouting holes.

[image]

kiqsmg, 2021-04-28:
Both options can work, but I still prefer the first one because it only requires to create more space between batteries and also helps to have a better fixation points (the closer the fixations are, the less vibration problems and bending possibility).

closed 2021-06-15 by YanAzer