spacelab-ufsc/battery-module-4c #12: Change the 8 pin PicoBlade vertical connector to the right angle version

YanAzer, 2020-10-03:
Current one is (0533980871)[https://www.digikey.com/en/products/detail/molex/0533980871/699072], for better compatibility this component should be changed to the right angle version (0532610871)[https://www.digikey.com/en/products/detail/molex/0532610871/699100].

andrempmattos, 2020-10-04:
The current version uses a 0532610871 (the right one) in the BOM file, but the 0533980871 in the board footprint/3D render. Then, it is important to update the footprint/3D render as suggested.

YanAzer, 2021-01-22:
@andrempmattos fixed solved issue via commits 626e60e940f685972dce35918c38ef1ae062296c and e58543fbc46f1a04be224ae19969c023d9c77b60, new outputs were generated via 2595ccffb678d09d9985b111dd0b83d6000be54a.

closed 2021-01-22 by YanAzer